ntu-aris/ntu_viral_dataset #1: Unzip error of 'sbs_03'

peterWon, 2021-04-14:
**Describe the BUG**
Thanks for your work! I'm using your dataset to test my SLAM approach.
All files can extract successfully except 'sbs_03'. I downloaded this file 4 times, the zipped file is with size about 2.9G. But when I unzip it, the error happens. I wonder maybe the uploaded file broken. 

**To Reproduce**
Just download and unzip. The error will appear.

**Site link or Repository**

brytsknguyen, 2021-04-14:
Hi Peter,

Thanks for trying out and notice us of the problem. I have replaced the sbs_03 file in the repository and updated the link. Please try downloading again!

peterWon, 2021-04-14:
Thanks for your quick reply!  I'll try it again.

closed 2021-04-15 by peterWon