ntu-aris/ntu_viral_dataset #2: Question abourt the vertical LiDAR's extrinsics

peterWon, 2021-04-14:
Thanks a lot for your work!

### Problem
Recently, I'm tring to test my SLAM method using your multi-lidar dataset. However, I find a problem in the provided calibration parameters. 
In detail, first, the last item of the vertical lidar's extrinsic should be 1, not 0.

Second, I transform the 'T_Body2Lidar' to roll, pitch and yaw, resulting in 1.57, 3.14, 0 respectively. I orginzed the two lidars and the IMU into one urdf file, and checked the projected clouds in rviz. But the result showing the pointcloud of the vertical lidar is upside down with the provided extrinsics. When I change its euler angles to '1.57, 0, 3.14', the result seems good.  I don't know where goes wrong.

[image]

### Reproduce
Here are related scirpts to reproduce the error. 
[tf_debug.zip](https://github.com/ntu-aris/ntu_viral_dataset/files/6308944/tf_debug.zip)

brytsknguyen, 2021-04-14:
Hi Peter,

How do you calculate the euler angles?

For the provided extrinsics [-1 0 0; 0 0 1; 0 1 0], the matlab command `rotm2eul([-1 0 0; 0 0 1; 0 1 0])` returns the yaw-pitch-roll sequence [3.14, 0, 1.57], which is indeed the result that gives you the correct aligment.

peterWon, 2021-04-14:
Thanks for your quick reply!
I resorted Eigen to convert the matrix to euler angles. I'll check my codes again.

peterWon, 2021-04-14:
@brytsknguyen After checking my codes, I confused Eigen's eulerAngles order. Thank you again. I'll close this issue.

closed 2021-04-14 by peterWon