ntu-aris/ntu_viral_dataset #4: Projecting pointclouds into camera image

cvsod, 2021-12-07:
Thanks for your great work!

I want to use the dataset for depth estimation, i wonder how can i project pointclouds into camera image?

cvsod, 2021-12-07:
> Thanks for your great work!
> 
> I want to use the dataset for depth estimation, i wonder how can i project pointclouds into camera image?

now, I do this through the following steps:
i.   obtain scans from the horizontal lidar;
ii.  transform the scan to the body system: Horizontal_scan = T_body2horizontalLidar * Horizontal_scan;
iii inverse T_body2cam -> T_cam2body
iii  transform the scan to the camera system: k(intrinsics matrix) * T_cam2body * Horizontal_scan;
iv: Visual point in camera image:
[image]

it seems wrong, cloud you point out the wrong step, thanks !

snakehaihai, 2021-12-09:
> > Thanks for your great work!
> > I want to use the dataset for depth estimation, i wonder how can i project pointclouds into camera image?
> 
> now, I do this through the following steps: i. obtain scans from the horizontal lidar; ii. transform the scan to the body system: Horizontal_scan = T_body2horizontalLidar * Horizontal_scan; iii inverse T_body2cam -> T_cam2body iii transform the scan to the camera system: k(intrinsics matrix) * T_cam2body * Horizontal_scan; iv: Visual points and corresponding depth in camera image: [image]
> 
> it seems wrong, cloud you point out the wrong step, thanks !

Hi. 
From your equation, it seems correct. Can you track this line and see its origin? it could be some data representation issue. This is a 16 line LIDAR. By right, you should see 16 horizontal lines. 

You may follow the sample here to test out the projection. We tested it and it does not have your issue. 

https://github.com/heethesh/lidar_camera_calibration

[image]

brytsknguyen, 2021-12-09:
Hi,

I think the problem is in your step iii. The procedure is as follows:

1. Converted the pointcloud to camera frame
2. Convert the coordinates to the normal coordinates (divide everything by z).
3. Apply distortion on the normal coordinates
4. Apply the intrinsic parameters to convert the coordinates to the pixel coordinates.

I have written some matlab scripts to demo this projection procedure at https://github.com/brytsknguyen/PointCloudProjection. I hope it works for your case.

cvsod, 2021-12-10:
> Hi,
>
> I think the problem is in your step iii. The procedure is as follows:
>
> 1. Converted the pointcloud to camera frame
> 2. Convert the coordinates to the normal coordinates (divide everything by z). [...]

thank you very much, it helps me a lot!

cvsod, 2021-12-10:
> Hi. From your equation, it seems correct. Can you track this line and see its origin? it could be some data representation issue. This is a 16 line LIDAR. By right, you should see 16 horizontal [...]

thanks! I get the right results!

closed 2021-12-10 by cvsod