ntu-aris/ntu_viral_dataset #5: Confusion about the horizontal LiDAR's extrinsics

cvsod, 2022-01-05:
Thanks for your great work!

I have some confusion about the process:
(i) T_Body_Cam in camera_left.yaml and
PointCloudProjection/project_pcd_to_img.m seems different, which is accurate?
(ii)  I visualized the depth values of some points of 0747.png in sbs_03:
[image]

There are some deviation，the horizontal Lidar's extrinsics seem not accurate, can you help me?
Hope for your reply! Thanks!

brytsknguyen, 2022-01-05:
Hi cvsod,

> (i) T_Body_Cam in camera_left.yaml and
> PointCloudProjection/project_pcd_to_img.m seems different, which is accurate?

The **T_Body_Cam** extrinsic in the **camera_left.yaml** is the correct one. I copied the wrong **T_Body_Cam** to into **project_pcd_to_img.m**. I have updated the repo [PointCloudProjection](https://github.com/brytsknguyen/PointCloudProjection).

However, after changing the extrinsic, I find that there the projection result is still similar to yours.

[image]

This is expected. There are two practical issues that affect the association of the image and the pointcloud.

First, the image and the pointcloud topics are not synchronized. If you check the time stamp of the image and pointcloud sample you can find that there is a 0.05s difference.
[image].

Second, the points in the pointcloud are also not captured at the exact time as in the pointcloud timestamp. The pointcloud accumulates all the points capture within a 0.1s. The timestamp of the pointcloud corresponds to the time when the last point in the pointcloud is captured.

These time errors will cause the pointcloud projection to misalign with the image, especially when the UAV's motion is aggressive.

To mitigate the first type of error one can use the camera timestamps to split and merge the pointclouds so that the pointcloud and image data are appropriately synchronized with respect to their timestamps (see Fig. 4 in [MILIOM](https://arxiv.org/pdf/2104.11888.pdf) and Fig. 3 in this [VIRAL](https://arxiv.org/pdf/2105.03296.pdf) papers). The second type of errors can be mitigated by using the motion distortion technique common in LIOM methods. These are doable approaches but I think they are quite an exhausive process.

cvsod, 2022-01-12:
Thanks for your reply.  I get it.

closed 2022-01-12 by cvsod