ntu-aris/ntu_viral_dataset #6: Any launch or yaml files work for LVI-SAM?

h-k8888, 2022-03-20:
I'm trying to do some experiments of LVI-SAM using NTU VIRAL dataset.
Any available launch or yaml files work for LVI-SAM?
Thanks.

h-k8888, 2022-03-20:
I tryed https://github.com/brytsknguyen/LVI-SAM.
The Image Feature seems right, but the Image Depth seems wrong.
Is this just a display problem？
[image]

brytsknguyen, 2022-03-20:
Hi,

Yes, I have been testing this method with NTU VIRAL dataset and the visual fusion works normally. It's just that the visua features points appear to flip to the back of the camera. I think it's a visualization issue on the LVI's implementation side where the author assumed some fixed transform between the camera and the lidar.

[image]

If you can figure out how to make the visualization consistent I'd love to know too.

h-k8888, 2022-03-20:
I tried. But as you said, there are some fixed transform in LVI-SAM, and the coordinate transformation is too complex. 
I haven't solved it yet.
 I'll let you know if it's solved.
Thanks.

h-k8888, 2022-04-02:
[image]
emmm.......
I think I solved this problem.
You need to modify a lot of code.

the extrinsic parameters from lidar to IMU, camera to IMU
in /src/visual_odometry/visual_estimator/utility/utility.h and /src/visual_odometry/visual_estimator/utility/visualization.cpp

and the lidar to camera frame
in /home/hk/ws_pl-lvi-sam/src/PL-LVI-SAM/src/visual_odometry/visual_feature/feature_tracker.h

closed 2022-05-23 by h-k8888

Stephen1e, 2022-12-12:
Hello, could you share the modified LVI-SAM code? It will be very helpful to me. Thank you in advance

sp225, 2023-11-14:
> [image] emmm....... I think I solved this problem. You need to modify a lot of code.
>
> the extrinsic parameters from lidar to IMU, camera to IMU in [...]

Hello, could you please share the modified code? It will be very helpful for me. Thank you very much.