ntu-aris/ntu_viral_dataset #7: Extrinsics between Lidar to IMU???

zuowei1998, 2022-04-16:
**Describe the BUG**
Could u tell me the Extrinsics between Lidar to IMU???
Matrix<3, 3> R
Vector3d T
Thanks a lot!

brytsknguyen, 2022-04-16:
Hi, which Lidar are you referring to?

When you download the dataset, the extrinsics are given in the yaml files:
[image]
[image]

For the horizontal lidar's extrinsics in the above, the lidar extrinsics are
Matrix<3, 3> R << 1, 0, 0, 0, 1, 0, 0, 0, 1;
Vector3d T << -0.05, 0, 0.055;

Note that the z axis of both the horizontal lidar and the IMU are pointing downward.
[image]

You can check out Section 4.4 in our paper for more information on the calibration.
https://github.com/ntu-aris/ntu_viral_dataset/blob/gh-pages/docs/NTU_VIRAL_Dataset_Submission.pdf

zuowei1998, 2022-04-17:
Thank u for your answer! That's helpful for me!
One more thing is that when I download dataset, always download failed. Maybe "Internet wall" in china, do u have some link like baiduNetpan or AliNetpan? 
sincerely thanks!!!

brytsknguyen, 2022-04-17:
Hi, does the download fail in the middle of the process or does it fail right when you start downloading?

Meanwhile, can you try downloading from NTU server below?
https://researchdata.ntu.edu.sg/dataset.xhtml?persistentId=doi:10.21979/N9/X39LEK

snakehaihai, 2022-04-17:
> Thank u for your answer! That's helpful for me! One more thing is that when I download dataset, always download failed. Maybe "Internet wall" in china, do u have some link like baiduNetpan or AliNetpan? sincerely thanks!!!

We baiduPan and Alipan are extremely slow to work with on our campus. I can try to upload, but cant promise. Are you able to use otherways like google drive or dropbox?

zuowei1998, 2022-04-18:
@brytsknguyen the download fail in the middle of the process. 
I have tried downloading from NTU server, but not work.

zuowei1998, 2022-04-18:
@snakehaihai  google drive and dropbox are restricted in china. It's terrible.

zuowei1998, 2022-04-18:
[image]

snakehaihai, 2022-04-18:
Baidupan tooks 16 years to upload as the upload speed is in kbs.     
I`ll upload to some other online drive and see it if can work. WIll post the new link here in about 2 hours

brytsknguyen, 2022-04-18:
Hi @zuowei1998, can you try downloading using wget in ubuntu:

`wget https://researchdata.ntu.edu.sg/api/access/datafile/68133`

The link for each file can be obtained by right-clicking and selecting "Copy link address" on the website:
[image]

snakehaihai, 2022-04-18:
> [image]

Hi This is the first link
Can you try if this service is workable in china?  

https://www.transfernow.net/dl/20220418j6WdhRxb

snakehaihai, 2022-04-18:
> wget

This is the second link service

https://send.tresorit.com/a#vfsO82cl177z-BIPiQ-hJw

Can you try if this works ?

zuowei1998, 2022-04-18:
@snakehaihai Both of them could be worked in china. But the second seem more better for free.hahaha~
I hope to get all of the dataset.Thanks~ 
@brytsknguyen I'm downloading from the command line. I don't know how it turns out. Good luck~

snakehaihai, 2022-04-18:
> @snakehaihai Both of them could be worked in china. But the second seem more better for free.hahaha~ I hope to get all of the dataset.Thanks~ @brytsknguyen I'm downloading from the command line. I don't know how it turns out. Good luck~

ok i`ll send you the rest of the link tmr.

Rgs
Shenghai Yuan

zuowei1998, 2022-04-18:
@brytsknguyen failed again. emmm maybe...
[image]

snakehaihai, 2022-04-19:
I`ll update the list here accordigly
EEE01
https://www.transfernow.net/dl/20220418j6WdhRxb
EEE02

EEE03
https://www.transfernow.net/dl/20220419a4H2rmzH/B9JXMKC2
NYA01

NYA02

NYA03

SBS01

SBS02

SBS03

Groundtruth Color 3D map

zuowei1998, 2022-04-19:
@snakehaihai Thank you very much. The download speed is very fast! Waiting for the remaining datasets to be updated.

snakehaihai, 2022-04-19:
@zuowei1998 
I uploaded all bags and some additional groundtruth.
You may download them now. 

I`ll try to find a way to make the transfernow link permanent. But for now, those links will last for 1 week. So try to download them all asap

I`ll close this issue for now. If you have any other issue, you may open new tickets.

snakehaihai, 2022-04-19:
@zuowei1998  
Can you help us to try if Microsoft cloud is allowed in china?
https://entuedu-my.sharepoint.com/:u:/g/personal/shyuan_staff_main_ntu_edu_sg/EbsL7sB6hbpLlqIoM2zkjHUBSQrgihrpq2_gN20sYveHJg?e=ALC8z8

zuowei1998, 2022-04-20:
@snakehaihai China support for Microsoft cloud. But the speed is about 70% slower than above ways.about 800~900kb/s

snakehaihai, 2022-04-20:
> @snakehaihai China support for Microsoft cloud. But the speed is about 70% slower than above ways.about 800~900kb/s

ok thx. Our school has free 1TB space for all student and staff with Microsoft cloud. We will be adding the microsfot link soon. Thanks for your help.

closed 2022-04-20 by snakehaihai