ntu-aris/ntu_viral_dataset #8: matlab script error!!!

zuowei1998, 2022-05-04:
[image]
do u know how to fix it??? thanks

brytsknguyen, 2022-05-04:
Hi  @zuowei1998 I just replied to this on the aloam repo
https://github.com/brytsknguyen/A-LOAM/issues/1#issuecomment-1117173414

brytsknguyen, 2022-05-04:
Copied from https://github.com/brytsknguyen/A-LOAM/issues/1#issuecomment-1117173414

You can comment out line 19 to 29, and line 32, and uncomment line 31 as follows:

% myCluster   = parcluster('local');
% 
% fprintf('Number of tests: %d. Number of Workers: %d\n\n',...
%          length(tests), myCluster.NumWorkers);
% 
% workers_needed = min(6, length(tests));
%      
% if myCluster.NumWorkers < workers_needed
%     delete(gcp);
%     parpool(workers_needed);
% end

for n=1:tests_count
% parfor n=1:tests_count
    
    [P_ate, rot_rmse] = check_viral_cal_err_parall(n, tests(n));

    EXP_NAME(n) = {tests(n).name(8:end)};
    P_ATE(n)    = {P_ate};
    R_ATE(n)    = {rot_rmse};
    
%     pause;
end
The commands from line 19 to 29 are to generate local worker pool to analyze the experiments in parallel to save time. The error may have something to do with your matlab subscription. Commenting them out make the analysis runs in sequence, which takes more time.

closed 2022-07-06 by brytsknguyen