ntu-aris/ntu_viral_dataset #11: Cannot parse the UWB data from rosbag.

zbabqr, 2022-07-06:
hello,
I want to parse the UWB data from the rosbag. However, when I use the topic name"/uwb_endorange_info", I get an error: (unicode error) 'utf-8' codec can't decode byte 0xa1 in position 1548: invalid start byte. Do I use the wrong topic name?

brytsknguyen, 2022-07-06:
Hi,

Can you try adding this to the beginning of the script?

```
import sys
reload(sys)
sys.setdefaultencoding("utf-8")
```

This is a commonly known issue with python and ros messages, not an error in our data.
https://stackoverflow.com/questions/3828723/why-should-we-not-use-sys-setdefaultencodingutf-8-in-a-py-script

brytsknguyen, 2022-07-06:
Hi,

Can you provide the code? Also which bag file may have issue?

Below is a short program to print out the data of messages on the the /uwb_endorange_info from the eee_01.bag

```
#!/usr/bin/python

import argparse
import sys
import os
import rospy
import rosbag

from uwb_driver.msg import UwbRange
from geometry_msgs.msg import TwistStamped
from geometry_msgs.msg import PoseStamped
from geometry_msgs.msg import PoseWithCovarianceStamped

# Some topics needs utf8 to decode
reload(sys)  
sys.setdefaultencoding('utf8')

if __name__ == '__main__':
   
    inbag = rosbag.Bag("/mnt/1FDD5C4564388F8F/DATASETS/NTU_VIRAL/eee_01/eee_01.bag",'r')

    for topic, msg, t in inbag.read_messages():
        if topic in ['/uwb_endorange_info']:
            print "topic:",  topic, " Stamp: ", msg.header.stamp.to_sec(), "Dis: ", msg.distance, "Panchor: ", msg.responder_location.x, msg.responder_location.y, msg.responder_location.z

```
You should see the print out like this

[image]

brytsknguyen, 2022-07-06:
It looks like you're missing the import of the uwb message definition. Can you git clone the package https://github.com/ntu-aris/uwb_driver to the workspace, catkin make / build the workspace, add `from uwb_driver.msg import UwbRange` to the beginning of python script, and try again?

Below is a short modification of your code.

```
import roslib
import rosbag
import rospy
import rosbag

from uwb_driver.msg import UwbRange # Install from https://github.com/ntu-aris/uwb_driver

bag = rosbag.Bag("/mnt/1FDD5C4564388F8F/DATASETS/NTU_VIRAL/eee_01/eee_01.bag",'r')
    
for topic, msg, t in bag.read_messages("/uwb_endorange_info"):
    timeUWB = "%.6f" %  msg.header.stamp.to_sec()
    requester_id = msg.requester_id
    responder_id = msg.responder_id
    uwbData = timeUWB + " " + str(requester_id) + " " + str(responder_id)
    print(uwbData)
```