ntu-aris/ntu_viral_dataset #13: outdoor wide scenario request

lyeemax, 2022-10-26:
Dear author, thanks a lot for your contribution on this dataset. I wonder if you can provide some records of wide scenario like forest, farm or something.

snakehaihai, 2022-10-26:
Hi, We are going to release a multi-campus dataset. Some place has forest-like environment. As for farms, we currently can't find such a place.

lyeemax, 2022-10-26:
Really appreciate!!!

closed 2022-10-26 by lyeemax

brytsknguyen, 2022-10-26:
Hi @lyeemax
As snakehaihai mentioned, we are working on some datasets at wider scales, there will be significant vegetation in the environment. Please look out for our updates in March/April.

[image]

[image]

lyeemax, 2022-10-27:
Awesome! I'm working on fusing lidar and camera SLAM on degenerated scenario, what I need is extract your wide and rgb-colored dataset ,I wanna thank you again for feeding data in my project and this apple Easter egg:
[image]
Waiting for your new release!