ntu-aris/ntu_viral_dataset #14: UWB anchor initialization

RicardoCzj, 2022-12-04:
Thank you so much for the excellent open source data set. We would like to know if it is possible to provide the 3D initial coordinates of UWB mobile antennas (e.g. 200A,200B,201A,201B) in the defined W-frame?

brytsknguyen, 2022-12-05:
Hi RicardoCzj,

Just to confirm, you are looking for the UAV ranging nodes' coordinates in the {W} frame defined by the UWB anchor nodes?

(image from the figure)
[image]

If so I think we can obtain it by using VIRAL SLAM pose estimate + UWB body-frame-referenced coordinates of the UWB ranging nodes. I may need to do some experiments and get back to you on it.

RicardoCzj, 2022-12-05:
Thank you very much for your reply. As you said, we want to know the 3-D position coordinates of the 4 UWB ranging nodes (200A,200B,201A,201B) under the defined {W} frame (the datasets we are currently concerned about are the rtp/nya/tnp datasets). If it is convenient, we also want to know what is the precision of the coordinates provided? Are there any ground truth values regarding the 4 ranging nodes?

abhigoudar, 2023-09-25:
Any updates on this?