ntu-aris/ntu_viral_dataset #15: building inspection datasets

lhthq, 2023-06-11:
Hi, I am working on the Lidar-Inertia-Ranging fusion. And I did not find building inspection datasets in the VIRAL website. Could you please provide me the building inspection datasets with groundtruth? Thanks a lot!

brytsknguyen, 2023-06-11:
Hi,

I am not sure if we can release the building inspection sequences as they were for some projects in collaboration with Singapore building and construction authority. @snakehaihai can confirm?

On the other hand, have you checked out the RTP and SPMS sequences? They are quite similar and may be more challenging than the building inspection sequences.

[image]

lhthq, 2023-06-11:
OK, thanks for replying. And the the upload sequences seem lacking of  rotation ground truth to evaluate the rotation error.

snakehaihai, 2023-06-12:
@lhthq Currently, we don't have permission to release the HDB building inspection dataset. But if you want to, I can help to ask for such permission. Where are you based, and what is the purpose of using such a dataset?

snakehaihai, 2023-11-30:
close the issue for inactive

closed 2023-11-30 by snakehaihai