ntu-aris/ntu_viral_dataset #19: radar drift by running lvi_sam

520sp, 2023-11-02:
Hello, when I use the eee_01 sequence to run lvi-sam, if I subscribe to "/os1_cloud_node2/points", the radar will drift significantly. If I subscribe to "/os_cloud_node_merged", no point cloud will appear in rviz, only the visual odometry. Please take a look at your side to see if such an error has occurred. If there is any improvement, I will be very grateful.

snakehaihai, 2023-11-02:
since when did we have radar...  u mean LIDAR?

brytsknguyen, 2023-11-02:
Hi, merging the pointclouds should be the issue of a software package. Why do you raise an issue at this dataset site?

brytsknguyen, 2023-11-02:
Hi @520sp , 

You can check out this package https://github.com/brytsknguyen/slict/blob/master/launch/run_ntuviral.launch

It has node that merges the lidar data into a single topic.

closed 2023-11-02 by brytsknguyen

520sp, 2023-11-04:
> Hi @520sp ,
> 
> You can check out this package https://github.com/brytsknguyen/slict/blob/master/launch/run_ntuviral.launch
> 
> It has node that merges the lidar data into a single topic.

OK, thank you very much for your response, I will try it.

sp225, 2023-11-14:
> Hi @520sp ,
> 
> You can check out this package https://github.com/brytsknguyen/slict/blob/master/launch/run_ntuviral.launch
> 
> It has node that merges the lidar data into a single topic.

Hello, sir, I have tried modifying the YAML file, but there is still a significant drift. I suspect that there might be some issues in the lvi-sam algorithm that are not compatible with this dataset, and we need to modify the code.