ntu-aris/ntu_viral_dataset #22: Request for the evaluation results of openvins in VIRAL dataset

ser666, 2024-01-24:
hello，
I do not find the results of openvins for in VIRAL dataset from the dataset paper. Using the code from https://github.com/brytsknguyen/open_vins, align using sim3，the results of the average ATE is about 2.This result is very different from those in other papers. Did I make an error in my assessment? What is the real result?
thank you.

snakehaihai, 2024-01-24:
Did you compensate the TF between Leica Prisim and IMU? it is one of the most common mistake we observe among all the user who use our dataset

ser666, 2024-01-24:
> Did you compensate the TF between Leica Prisim and IMU? it is one of the most common mistake we observe among all the user who use our dataset
thank you,I will try again

ser666, 2024-01-24:
> Did you compensate the TF between Leica Prisim and IMU? it is one of the most common mistake we observe among all the user who use our dataset

thank you, i will try again

ser666, 2024-02-05:
> Did you compensate the TF between Leica Prisim and IMU? it is one of the most common mistake we observe among all the user who use our dataset

hello，the openvins test results obtained based on the provided matlab code which compensated the TF are as follows. Is this consistent with your correct result?
```
Number of tests: 9. Number of Workers: 10

test:  2. eee_02_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	2.5456	  5.9785.

test:  1. eee_01_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	5.2512	  5.6739.

test:  4. nya_01_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	1.2816	  7.7022.

test:  8. sbs_02_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	4.6721	  7.3288.

test:  6. nya_03_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	0.3091	  3.1442.

test:  7. sbs_01_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	3.0734	  5.3498.

test:  3. eee_03_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	4.6288	 10.8618.

test:  5. nya_02_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	0.3539	  4.8083.

test:  9. sbs_03_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	5.2861	 25.8650.

RESULT =

  9×3 cell array

    {'eee_01_0anc_vis'}    {[5.2512]}    {[ 5.6739]}
    {'eee_02_0anc_vis'}    {[2.5456]}    {[ 5.9785]}
    {'eee_03_0anc_vis'}    {[4.6288]}    {[10.8618]}
    {'nya_01_0anc_vis'}    {[1.2816]}    {[ 7.7022]}
    {'nya_02_0anc_vis'}    {[0.3539]}    {[ 4.8083]}
    {'nya_03_0anc_vis'}    {[0.3091]}    {[ 3.1442]}
    {'sbs_01_0anc_vis'}    {[3.0734]}    {[ 5.3498]}
    {'sbs_02_0anc_vis'}    {[4.6721]}    {[ 7.3288]}
    {'sbs_03_0anc_vis'}    {[5.2861]}    {[25.8650]}

Elapsed time is 18.399795 seconds.
IdleTimeout has been reached.
Parallel pool using the 'local' profile is shutting down.
>> checkall_parallel
Number of tests: 9. Number of Workers: 10

Starting parallel pool (parpool) using the 'local' profile ...
Connected to the parallel pool (number of workers: 10).
test:  3. eee_03_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	4.6288	 10.8618.

test:  2. eee_02_0anc_vis.
Error:	Pos [m]	 Rot [deg]
[... 130 lines cut ...]
test:  4. nya_01_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	1.2816	  7.7022.

test:  5. nya_02_0anc_vis.
Error:	Pos [m]	 Rot [deg]
OVODOM:	0.3539	  4.8083.

RESULT =

  9×3 cell array

    {'eee_01_0anc_vis'}    {[5.2512]}    {[ 5.6739]}
    {'eee_02_0anc_vis'}    {[2.5456]}    {[ 5.9785]}
    {'eee_03_0anc_vis'}    {[4.6288]}    {[10.8618]}
    {'nya_01_0anc_vis'}    {[1.2816]}    {[ 7.7022]}
    {'nya_02_0anc_vis'}    {[0.3539]}    {[ 4.8083]}
    {'nya_03_0anc_vis'}    {[0.3091]}    {[ 3.1442]}
    {'sbs_01_0anc_vis'}    {[3.0734]}    {[ 5.3498]}
    {'sbs_02_0anc_vis'}    {[4.6721]}    {[ 7.3288]}
    {'sbs_03_0anc_vis'}    {[5.2861]}    {[25.8650]}

Elapsed time is 20.749150 seconds.
```
thank you for your reply

snakehaihai, 2024-02-05:
Hi,

On average, the OPENVINS results we obtain fall between 0.5 and 1. Some of your data is better than ours, with an APE of 0.3 for two sequences, while some are not. If you have already compensated for the transformation (TF) between the sensor and Ground Truth, it is likely that minor errors are caused by parameters, eign library, or other factors.

If you have concerns about this in your publication, you may consider changing the library or fine-tuning parameters to get a better result. But, often it is not necessary.

As long as others can replicate your experiment using the settings you provided and you are using the same parameters across all sequences, I don't believe it will be a major issue in publications (since you have evidence).

In publications, we have observed different people obtaining different results with the same dataset.

Best regards,
Shenghai Yuan