ntu-aris/ntu_viral_dataset #25: Someting wrong  extrinsic parameters

Lucian011409, 2024-12-04:
Hello, I would like to express my sincere gratitude for your work. However, I have encountered some issues: when using the LIO-SAM algorithm and playing the .bag file, the RViz interface keeps shaking vertically. Although I used the script you provided to fix the IMU timestamp issue, it did not resolve the problem. Could it be that there is an error in the IMU to LiDAR extrinsic parameters? My IMU extrinsic parameters are: extrinsicTrans: [0.050, 0.000, -0.055]. I look forward to your reply.

brytsknguyen, 2024-12-04:
Hi,

Can you try out my fork here?

The extrinsic settings for one lidar is here
https://github.com/brytsknguyen/LIO-SAM/blob/master/config/params_ntuviral_horz.yaml

LIO-SAM is quite crude in its handling of IMU and rotation stuff. I remember the gtsam by default uses some IMU preintegration implementation that has issues when IMU has the z axis pointing downwards. So you must build gtsam with a specific flag to disable that IMU preintegration method as mentioned here:

https://github.com/TixiaoShan/LIO-SAM/issues/247#issuecomment-849257914

Lucian011409, 2024-12-05:
Thank you very much for your patient reply. In fact, this is my first time asking a question on Github, and your thoughtful response has been a great encouragement to me as a beginner. Once again, thank you, and I will take your advice to try and resolve the issue.

closed 2024-12-16 by snakehaihai