ntu-aris/ntu_viral_dataset #28: Missing pose ground truth in spms02

KevinJohnsonL, 2025-03-11:
Hi, thanks for your dataset shared!
I was testing my Lidar odometry with ntu viral dataset, APE evalution by evo.  The error increases quickly in  runtime: 265 -> 270 s. Then I found out GROUND TRUTH MISSING in spms_02 during timestamp:
1624181364.522364049 -> 1624181385.672683980 id: 4379 ->4380 21s

In evo, it seems just interpolate poses when grouond truth missing. 

[image]

snakehaihai, 2025-03-12:
Hi, I believe this is the period where the Leica loses track of the prism, meaning no valid data can be generated. I suggest interpolating the pose to maintain continuity for EVO when calculating the APE, but excluding the interpolated sections from the final APE calculation to ensure accuracy.

closed 2025-03-12 by snakehaihai

brytsknguyen, 2025-03-12:
Hi @KevinJohnsonL 

The Leica tracker loses track in some periods so there are no ground truth data in those periods.

Did u try the official [script](https://ntu-aris.github.io/ntu_viral_dataset/evaluation_tutorial.html#jupyter-notebook-released) to calculate the error? It resamples the gtr data based on the timestamps so the period without gtr data will not be counted into the RMSE.