ntu-aris/ntu_viral_dataset #30: NTU Viral Result OF FastLIO2

Uintm, 2025-06-17:
Based on the FastLIO2 results you provided, they are inconsistent when compared to the results in the Traj-LO paper.

brytsknguyen, 2025-06-17:
There can be some differences based on the settings chosen by each paper. However I think the difference is small.

Uintm, 2025-06-17:
Thank you for your prompt reply. The results in traj-LO differ significantly from the sample results you provided.

[image]

brytsknguyen, 2025-06-18:
Hi,

I am not sure how Zheng tuned the parameters to get such high accuracy for the L1 + I setup. Maybe you can contact him.

Below is the result for the Fast-LIO2 with 1 lidar + IMU in the "eigen is all you need" paper.

[image]

The params for this experiment can be found at 

https://github.com/brytsknguyen/FAST_LIO/blob/master/config/ntuviral.yaml